GEIGEIGEIST/TOTEM #1: 36 keys version

toniz4, 2022-11-10:
Hello! This is a awesome design, the only problem for me is the outer pinkie key. 

It would be easy for me to modify the pcb file to remove that key, the problem is in the case files. I can only barely use freecad, and by what I saw, there is only the stl and step files avaliable. 

So, it's possible to release a 36 key version of the TOTEM in the future? 

Thanks!

GEIGEIGEIST, 2022-11-10:
Hi Toniz4,
Unfortunately I don't plan to do different layouts with this board. I'll upload the PCB files today. And I can upload the Fusion360 files too, so people can modify it themself.

toniz4, 2022-11-10:
That's fine, Either way I'm building one for myself, that's a really nice design and it's nice to try other layouts. Just out of curiosity, in your personal layout, what key do you bind that outer pinky key?

Another unrelated question, what is the model number of the battery that you used? The biggest battery that i found is a 601417 with 150mah (https://a.aliexpress.com/_mqxZTWy). Maybe you managed to find a better one.

GEIGEIGEIST, 2022-11-10:
Thank you. I've added the pinky key to also support layouts where you use the pinky key for the letter usually on the top left and top right like the [Balbuzzard](https://github.com/brow/balbuzard). 

I use a 751517 from Walkera (100mah) after never receiving [this one](https://aliexpress.com/item/1005001812754397.html) I ordered at AliExpress (the problem was the logistics company)
Unfortunately I can't take a look at the one you've linked, cause they probably doesn't ship to Germany.

toniz4, 2022-11-10:
Thanks for the info!

closed 2022-11-10 by toniz4