GEIGEIGEIST/TOTEM #4: Case version

tarneaux, 2023-01-05:
Hi, I am planning to build this keyboard myself, but I saw that there was a totem artwork on the top of the case. I like it all to be minimalist, so is it possible to have a version of the cases without that artwork?
I'd love to do it myself if my freeCAD wasn't throwing a dependency error, and I'm also not quite sure I'd get the removal right.
Thanks

GEIGEIGEIST, 2023-01-08:
Oh rightwise you could totally do it, but I could send you a version without the artwork. I just don't know how to send it to you.

tarneaux, 2023-01-08:
Well actually I don't think I'm the only one who'd like this, so maybe a separate branch or something would be more appropriate. But if you don't want to add those versions to the repo, a wetransfer or something similar would be alright.
I was actually trying to do this myself before you answered, because I managed to fix the freeCAD issue, but I guess I don't really know how to do it. I tried a few different boolean operations but the artwork was still (roughly) there.

closed 2023-01-10 by tarneaux