GEIGEIGEIST/TOTEM #6: Tenting support

ar-qun, 2023-01-31:
Hello GEIGEIGEIST,

I really liked you keyboard and was wondering if tenting is possible with this keyboard?

Would be interesting to use it with Manfrotto tripods.

Also would it be possible to buy a pre-build version?

GEIGEIGEIST, 2023-01-31:
Hi ar-qun,
Unfortunately tenting isn't possible, since I couldn't fit the footprint for the SplitKB puck in my PCB, without making to many compromises.
A pre built version isn't planed, but at least a kit will be available soon.

dlehman, 2023-01-31:
@GEIGEIGEIST Any idea what timeframe for kit will be? Considering ordering my own parts, but would be willing to wait a bit for a kit.

closed 2023-02-03 by ar-qun

Wanderwichtl, 2023-02-06:
@GEIGEIGEIST I am currently thinking about building a copy of your awesome Totem. Is it worth it to wait for the kit, and what will it include?
Would it be possible to make it to a wireless version on my own?
Will it include a (transparent) case? 
Will it be a standard PBC?