GEIGEIGEIST/TOTEM #14: Recommendations on 3D Case and Information on Material Safety and Surface Finish

sedghi, 2023-10-10:
Thank you for sharing this excellent build. I have made the decision to proceed with building one myself. Your recommendations regarding the PCB options, such as the number of layers and thickness, were truly helpful.

I would greatly appreciate it if you could provide some recommendations for the 3D case as well. There seems to be an overwhelming number of options available, and I am unsure about which one to choose. I have come across information suggesting that resin can be toxic. Is this true for all materials used in the case, or are there some alternatives that are safer? Additionally, I would like to know more about the surface finish.

By the way, I really like the case design that you featured in the image. Could you please share the options that were used to generate it in JLCPCB?

[image]

spamwax, 2023-10-12:
I used pcbway.com and selected "Resin Standard white material (UTR 8360)", and it looks and feels great. I am not sure about toxicity of it!

spamwax, 2023-10-13:
No, I used pcbway because the other one didn't accept the file I was uploading

ivandaho, 2024-09-01:
Leaving this here in case someone else finds it useful - I also had difficulty deciding the options for the case.

These settings worked for me for printing the case via JLCPCB, with the exception of a small chip on the left half, near the thumb cluster breaking off. It's not critical and i was contacted by them about it.

[image]

 [image]