GEIGEIGEIST/TOTEM #16: Question about case compatibility?

jittapont, 2023-10-21:
I want to ask about the case compatibility between the USB version and BLE version. Can I use the USB version (https://github.com/GEIGEIGEIST/TOTEM/tree/main/case/USB) first and change to wireless option with the same case?
Thanks a lot

spamwax, 2023-10-22:
I believe it is compatible. Just make sure you disconnect the TRS cable between the two halves when you install the battery.

isopix, 2023-10-30:
What happens if we forgot to disconnect it? Can some components be damaged?

spamwax, 2023-10-30:
> What happens if we forgot to disconnect it? Can some components be damaged?

Yes, according to the build [guide](https://github.com/GEIGEIGEIST/TOTEM/blob/main/docs/buildguide.md#trrs-jacks-optional), it can damage it.