GEIGEIGEIST/TOTEM #17: https://keeb.supply/products/geist-totem doesn't work - We couldn't find any product at that address!

isopix, 2023-10-30:
https://keeb.supply/products/geist-totem doesn't work - We couldn't find any product at that address!

And I live in EU(Poland)

lemcz, 2023-11-06:
@isopix I also live in Poland, HMU if you're interested in buying a PCB (I can assemble it for you). I also ordered 3d printed case from jlcpcb, so I can help out with that process.

Other than that - this is not an issue for the author of the keyboard, but store's and AFAIK (I've asked on their discord), they are not going to re-stock on TOTEM any time soon.

isopix, 2023-11-06:
> @isopix I also live in Poland, HMU if you're interested in buying a PCB (I can assemble it for you). I also ordered 3d printed case from jlcpcb, so I can help out with that process.
>
> Other than that [...]

It's awesome looking and most compact keyboard. If some trackpad/trackpoint (or other pointing device) would be added, then it would be perfect travel keyboard (there are already keyboard with slim choc switches and custom trackpoint assemblies)

Currently I have very limited budget, and now I'm much more interested in wireless Charybdis:
https://bastardkb.com/product/charybdis-nano-premium-diy-kit/
https://github.com/Bastardkb/Charybdis

I haven't found anyone in EU(except official shop, where this premium set costs 800eur, but it's not even wireless [so it needs different controller, trackball optical sensor and PCBs]) that would sell me PCBs (especially wireless version)
There are also other mods, including 2nd trackball, encoders, etc.
It's like nylon prints costs 100eur for pair, the same for four metal plates (in Netherland, can only hope in Poland it's cheaper).
This are prices from jlcpcb, that makes PCBs and nylon prints (that are sanded too. BastardKB also orders from them, so it would look identical)

I never ordered anything 3d printed or jlcpcb and don't know metal laser-cutting prices in PL. I've simply contacted all the people that were hanging in discord server, that made wireless version, and everyone  replied that he sold his spare boards already(I've heard it's way cheaper to order multiple PCBs at once and that one set is not viable).

I didn't had enough pressure to start ordering it by myself, because there are improvements going all the time(especially as mods/forks).

Anyway feel free to contact me at krzysztof dot krakowiak at some popular 'G' provider, because Github doesn't allow direct messaging ;-)