GEIGEIGEIST/TOTEM #18: Issue with Totem Keyboard: Few Keys Not Working After a Few Days

marczen, 2023-10-31:
## Description:
Hello,

I'm a beginner in keyboard building and recently assembled my Totem keyboard. I've been enjoying it, but I encountered an issue where certain keys stopped working after a few days of use. I've attached an image to illustrate the problem.

## Problem Details:
The issue has persisted despite my attempts to troubleshoot, which included checking switches and hotswap sockets.
I'm not sure if the problem might be related to diodes, or why specifically these three keys are affected.
I would greatly appreciate any guidance or ideas on what might be causing this issue and how to resolve it.

Thank you for your assistance!

*(image of keys that don't work)*
[image]

dedebenui, 2023-11-16:
Hi! I'm also a beginner. I built the wireless Totem a 2 weeks ago and the left half (just like you!) had a few keys not working. I measured a bunch of paths with my multimeter in diode mode and noticed that I had some voltage drop across a diode even in reverse. I desoldered every diode and socket on the left board and could still measure a voltage drop across the pads of the faulty diode even though nothing was connected there, so it wasn't the diode's fault. Since I had spare boards because of the minimum order quantity, I soldered everything again on a new one and it still doesn't work: some keys don't work at all and some keys, like my `t` key sometimes randomly outputs a few adjacent letters in one press.

I had no issue with the right half.

I suspect that I make a mistake when I solder the controller and, because it's a difficult part to solder, I leave my iron too long on the contact point or press too hard, or something else. The result is a damaged trace on the board that sort of shorts 2 traces together. If I measure the resistance on a bare controller or on my working half between C0 and R0, I get >10MOhms while I get barely half that on the problematic one. I will try to solder a new controller onto a new board, be extra careful and report back.

If by any chance someone who knows what they're doing has some insight, it would be greatly appreciated 😄

skull141209, 2023-11-17:
LoL I've just have exactly problem with my new build

kubroid, 2023-12-21:
Same issue with the left RC1-1 key :(

saseba, 2024-09-27:
I had the same issue on a Build. Culprit was most likely Flux residues underneath the Controller/Between Pins.

I cleaned the Board using Isopropyl Alcohol (99%). Underneath controller is tricky, so I tented th PCB and used Isopropyl allow the Flux to flow downward. After that I gently heated the Area of the Controller (guess 60°C) to get the Isopropyl to evaporate and stop the Flux from causing shorts again.

For next Build I recommend to use short diode legs to connect Controller and PCB so you could use less solder (equals less flux)  to to better Heattransfer between the Pads. At least that's my impression.