GEIGEIGEIST/TOTEM #19: right hand board unresponsive unless flashed with left hand side FW

boogaloodude, 2024-04-07:
I am using the reference ZMK FW.  Left side board works great.  Right hand board is unresponsive unless flashed with the left side FW.  Is this an indication that I've somehow bungled the build?  Or an actual issue with the reference ZMK FW?

Cirius1792, 2024-12-12:
Same here, any hint on a solution?

Cirius1792, 2024-12-12:
Right after the comment I arrived to the obvious solution myself. 

The left part of the keyboard is the main one. The right side communicate through the left side. So having both halves powered on at the same everything works fine. 

The reason why I discovered it so late is because none of my halves work on battery power... but this is another story

burgessa23, 2025-01-24:
I'm having an issue where the right side won't work on battery. What was the problem with yours?

Cirius1792, 2025-01-24:
I forgot to solder the pins of the power switches 🙃

isakbosman, 2025-03-23:
> I am using the reference ZMK FW. Left side board works great. Right hand board is unresponsive unless flashed with the left side FW. Is this an indication that I've somehow bungled the build? Or an actual issue with the reference ZMK FW?

@boogaloodude In a wireless setup with ZMK, that is standard because the left side acts as the central in the bonding process unless you use a dongle.

isakbosman, 2025-03-23:
> I'm having an issue where the right side won't work on battery. What was the problem with yours?

Use a multimeter to test the GND and VCC when on battery only. If you get the expected voltage (3.3V) then its often the power switch like @Cirius1792  reported, otherwise its often the pads on the back (specifically B+). Just try to resolder those better