GEIGEIGEIST/TOTEM #25: Seeed Studio Fusion PCB Issues

Endracion, 2024-08-09:
After downloading the zip file and uploading it to their system, it gives the following errors:

Drill file not found. Please check that you have included the NC drill file in Excellon format. [Help](http://support.seeedstudio.com/knowledgebase/articles/1977138)

Outline not found. Please ensure you have an outline drawn in one of the files, ideally in a dedicated outline layer and that this layer is named appropriately (.GML, outline etc.) [Help](http://support.seeedstudio.com/knowledgebase/articles/1977141)

Parameters could not be extracted. Please try again or enter your board parameters manually. You can still add them to the cart but some automatic features will be restricted. [More info](http://support.seeedstudio.com/knowledgebase/articles/1977135)

--

So more importantly than the sizing, the Drill and Outline file don't seem to be compatible? Could there be an updated version?

Thanks.

that-ambuj, 2024-12-02:
Could try opening the .kicad file in Kicad Pro and exporting to gerber with the options to include drill files? Also look for related options to export the outline.

endikallanomatxin, 2024-12-27:
Same issue here.