GEIGEIGEIST/TOTEM #27: Case Step/STL files aren't recognisable by JLC3DP

Sarteg, 2024-09-02:
I tried to upload the step/stl files to JLC3DP to have the case printed but keep getting error.
Their support person tried to open the files directly using their software but failed with the following error:
[image]

I am not sure where the problem is was hoping if someone can take a look?

Kcebnelloh, 2024-11-27:
You don't need any software for ordering from jlc3dp. Just visit this [website](https://jlc3dp.com/3d-printing-quote) and drag and drop the files you want to print.

gienub, 2025-01-26:
This is still an issue using their upload tool on the website. Any pointers on a possible fix?

DancingSpell, 2025-07-30:
I had the same problem when downloading the step/stl files individually and trying to upload to jlcpcb (or indeed open for viewing/editing). However problem did not occur when I downloaded the entire repository as a zip file, and then dragged the step files direct from the zip into jlcpcb web interface. Hope this works for you too.