GEIGEIGEIST/TOTEM #36: Which Seeed XIAO board version to use for wireless build?

that-ambuj, 2024-11-22:
I was looking to buy the micro-controller to build this keyboard and I see two options for Seeed XIAO BLE on the website I usually order from:

1. [Seeed Studio XIAO ESP32S3 – 2.4GHz WiFi, BLE 5.0](https://robu.in/product/seeed-studio-xiao-esp32s3-2-4ghz-wifi-ble-5-0/) - This is a cheaper version with an ESP32 chip I assume.
2. [Seeed Studio XIAO BLE nRF52840](https://robu.in/product/seeed-studio-xiao-ble-nrf52840/) - This is double the price of the previous option with a different chip.

Please help me decide because I don't buy the wrong part.

that-ambuj, 2024-11-22:
P.S.: There are other versions available such as the [SAMD21 version](https://robu.in/product/seeedstudio-seeeduino-xiao-pre-soldered/) available on the website. See the search results here: https://robu.in/?s=seeed+xiao&post_type=product&dgwt_wcas=1 and LMK if you think one of them is useful.

Kcebnelloh, 2024-11-27:
From what I understand you have to go with the Seeed Studio Xiao BLE nRF52840. I am not sure about this tho. I am building the wired version right now and I initially bough the wrong microcontroller (esp based) and it was impossible to flash.

that-ambuj, 2024-11-27:
@Kcebnelloh Yeah, I checked ZMK's docs for flashing the SEEED XIAO and they barely mention the ESP32 version. Also, I saw an open issue for adding support for ESP32 during my research; so I figure nrf52080 must be the one. 

Thanks for the help!

closed 2024-12-02 by that-ambuj