GEIGEIGEIST/TOTEM #40: Qmk firmware flashing

Moholic, 2024-12-20:
Tried to flash the firmware through vial-qmk but wasnt able to make it work. Seemed to flash the layout of the left half to both sides.

Any help appreciated.

Moholic, 2024-12-23:
Ok, so I tried around a bit...
loading the precompiled firmware was sucessful - giving me only 4 of the 5 keyboard layers (adjust is missing)
Compiling via qmk works for the default, not for the vial enabled version.
Flashing the compiled version would flash me the left side on both halfs (mirrored on right).
Was trying to flash with the following command:
- qmk flash -kb geigeigeist/totem -km default
- qmk flash -kb geigeigeist/totem -km default -bl uf2-split-left
and
- qmk flash -kb geigeigeist/totem -km default -bl uf2-split-right
respectively.

Anyone able to help on the correct command? or am I just stuck with the precompiled firmware?

JMulder99, 2024-12-31:
Hi,
I've been trying to flash using WSL (Ubuntu). It compiles successfully for the default (vial enabled) and my custom keymap. 

However, when I try to flash the keyboard, it is stuck at "Waiting for the drive to deploy". Does anyone know is missing?

Moholic, 2024-12-31:
Hi,
Just press the reset-button twice and the device mounts as an external drive.

JMulder99, 2025-01-03:
I got it to work without the flash command, because WSL fails to recognize the drive.

- I added `split_keyboard = yes` in `rules.mk`. 
- I removed `reverse_hands = true` in `keyboard.json`. It causes the master to flip to RIGHT even though I set it to LEFT. I have no idea why that was put in. 
- Then I presses the reset twice, and drag&dropped the same `.uf2` file on both halves.

Kcebnelloh, 2026-09-17:
> Hi, Just press the reset-button twice and the device mounts as an external drive.
> 
> Btw: I had to add 'SPLIT_KEYBOARD = yes' to rules.mk. and then flashed it with:
> 
> * qmk flash -kb geigeigeist/totem -km default -bl uf2-split-left
>   and
> * qmk flash -kb geigeigeist/totem -km default -bl uf2-split-right
>   respectively.

Thanks a lot for this ❤️ Was stuck on this for ages!