GEIGEIGEIST/TOTEM #45: ZMK Flash, drag and drop causes drive to disconnect and reset files.

Chris-LewisI, 2025-01-23:
Every time I try to glad my Xiao with the ZMK UF2 for my Bluetooth set up the drive disconnects.

When I reopen the drive it has the original files all over again even if I deleted them previously.

isakbosman, 2025-03-23:
I'm not sure if this is still helpful or if it can help someone else. This is standard behavior as the MCU restarts once the firmware file uploads. The best thing to do is to either hit the reset button afterwards, or unplug and plug the left side in again. Try to type to see if the keys work. Do the same for the right side (it doesn't work without the left side *depending on your setup*)

pipkininthegrass-beep, 2026-01-29:
> I've tried plenty of advice and have never managed to fix this. If you discover anything let me know.

Most unfortunately, it turned out that shutting down the computer, waiting overnight, and starting it all up again (including reconnecting with bluetooth) proved that I did manage to successfully flash the uf2.  Very unscientific and unreplicable; 100% luck based.