GEIGEIGEIST/TOTEM #53: where is the laters zmk source code?

SpyrosRoum, 2025-04-21:
Hi, as per the title, I'm not sure where I can find the latest zmk version of the firmware. I hope I'm not being too blind here, thank you

frtytw, 2025-04-26:
https://github.com/GEIGEIGEIST/zmk-config-totem

SpyrosRoum, 2025-04-27:
This is supposed to be outdated according to the "Firmware" section of the readme in this repo.

frtytw, 2025-04-27:
I don't see that written anywhere. 
If you are referring to the commit dates, I don't think that matters.
If I understand correctly, the action should pull from the ZMK source repo?

iamavolk, 2025-06-23:
> I don't see that written anywhere. If you are referring to the commit dates, I don't think that matters. If I understand correctly, the action should pull from the ZMK source repo?

It is "outdated" according to _this_ repo's README. Any elaboration would be of great help