GEIGEIGEIST/TOTEM #55: M2 standoff cannot be inserted into the hole

chengyuancong, 2025-04-27:
Diameter of holes in the top case is about 3.2mm but diameter of M2 standoff is about 3.5mm. I tried m2 standoff from different sellers but none of them can be inserted. Did I do sth wrong?

frtytw, 2025-04-27:
Mine seated snugly with a bit of pressure. 
They need to be snug, in order not to fall out when assembling.

metzger100, 2025-06-05:
Is it too obvious, or is a drill not the right tool for this problem? I noticed the issue as well and still ordered 4 mm standoffs because I thought I could widen the holes with a drill (parts didn't arrive yet so I don't know if drilling works). With a resin print that shouldn’t be an issue, and even with a filament print you should be able to enlarge the holes with a drill.

sinefabula, 2025-06-05:
@metzger100 yes, a drill should definitely do the job. I just think it should not be a part of the regular build process when the case is 3D printed and can be easily modified to take whatever parts are readily available, not vice versa. Thanks for the drilling tips though, didn't know that, and that is helpful!

metzger100, 2025-06-05:
@sinefabula 

> to take whatever parts are readily available

Exactly. If the holes for the standoffs are to small you can change them with a drill. If they are to big, you are in trouble (fill it with hot glue and redrill? Not everyone has hot glue). I guess everyone has a drill at home. The too small holes are better in my opinion.

metzger100, 2025-06-13:
Okay. Feedback time. 

I used a 3.5mm and a 4mm drill to widen the holes. It worked, but there are a few problems with this method. 

1. You will probably destroy the countersinks for the screws on the top. But this is not a big deal. You use the screws on the top to cover the holes. So visually it doesn't make much of a difference (in one hole the screw is not centered because the widening has shifted the hole, so a small gap is visible). I had to use a small hammer to get the 4mm diameter standoffs into the widened holes so they are strong enough to hold without the screw on top. If you want to be sure, you can use super glue on the standoffs to glue them into the holes.
2. Also in my case the walls of the socket for the standoffs became quite thin. I used the JLC Black Resin which has good values for this kind of stress: Elongation at break, impact strength, tensile strength (higher is better). I am not sure if the Imagine Black resin could withstand this drilling. So be carefull

Lessons Learned:
1. It would be better to have the sockets for the standoffs on the bottom parts of the case so you don't have to drill into the top parts, which is the visible part of the keyboard.
2. It would be better if the standoff sockets were thicker so you could widen the holes without risking breaking the walls of the socket. Especially if you use worse resins.