GEIGEIGEIST/TOTEM #56: unable to configure keymap in via / vial + several other issues

X4ndras, 2025-05-07:
Hi there,

While setting up the Totem keyboard, i encountered several issues.

First, like in issue #40, I had to manually add `SPLIT\_KEYBOARD = yes` in the `rules.mk` file. It might be helpful to mention that in the docs.

Also, I couldn’t find clear instructions on how to flash the firmware if I missed something, feel free to point me there.

When trying to compile with `qmk compile -kb geigeigeist/totem -km vial`, I ran into an error:
```
➜  totem git:(master) ✗ qmk compile -kb geigeigeist/totem -km vial
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=geigeigeist/totem KEYMAP=vial KEYBOARD_FILESAFE=geigeigeist_totem TARGET=geigeigeist_totem_vial VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc (Arch Repository) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   38324       0   38324    95b4 geigeigeist_totem_vial.uf2

Compiling: .build/obj_geigeigeist_totem_vial/src/community_modules.c                                [OK]
Compiling: keyboards/geigeigeist/totem/totem.c                                                      [OK]
Compiling: .build/obj_geigeigeist_totem_vial/src/default_keyboard.c                                 [OK]
Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/eeconfig.h:24,
                 from quantum/keycode_config.h:23,
                 from quantum/quantum.h:41,
                 from ./.build/obj_geigeigeist_totem_vial/src/default_keyboard.h:28,
                 from ./keyboards/geigeigeist/totem/keymaps/vial/keymap.c:14,
                 from quantum/keymap_introspection.c:9:
quantum/keymap_introspection.c: In function 'tap_dance_count_raw':
quantum/keymap_introspection.c:129:23: error: 'tap_dance_actions' undeclared (first use in this function); did you mean 'tap_dance_action_t'?
  129 |     return ARRAY_SIZE(tap_dance_actions);
      |                       ^~~~~~~~~~~~~~~~~
quantum/util.h:37:68: note: in definition of macro 'IS_ARRAY'
   37 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:129:12: note: in expansion of macro 'ARRAY_SIZE'
  129 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/keymap_introspection.c:129:23: note: each undeclared identifier is reported only once for each function it appears in
  129 |     return ARRAY_SIZE(tap_dance_actions);
      |                       ^~~~~~~~~~~~~~~~~
quantum/util.h:37:68: note: in definition of macro 'IS_ARRAY'
   37 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:129:12: note: in expansion of macro 'ARRAY_SIZE'
  129 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/util.h:48:32: error: first argument to '__builtin_choose_expr' not a constant
   48 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:129:12: note: in expansion of macro 'ARRAY_SIZE'
  129 |     return ARRAY_SIZE(tap_dance_actions);
      |            ^~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:136:27: error: 'tap_dance_actions' undeclared here (not in a function); did you mean 'tap_dance_action_t'?
  136 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                           ^~~~~~~~~~~~~~~~~
quantum/util.h:37:68: note: in definition of macro 'IS_ARRAY'
   37 | #    define IS_ARRAY(value) (!__builtin_types_compatible_p(typeof((value)), typeof(&(value)[0])))
      |                                                                    ^~~~~
quantum/keymap_introspection.c:136:16: note: in expansion of macro 'ARRAY_SIZE'
  136 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/util.h:48:32: error: first argument to '__builtin_choose_expr' not a constant
   48 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:136:16: note: in expansion of macro 'ARRAY_SIZE'
  136 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/util.h:48:31: error: expression in static assertion is not an integer
   48 | #    define ARRAY_SIZE(array) (__builtin_choose_expr(IS_ARRAY((array)), sizeof((array)) / sizeof((array)[0]), (void)0))
      |                               ^
quantum/keymap_introspection.c:136:16: note: in expansion of macro 'ARRAY_SIZE'
  136 | _Static_assert(ARRAY_SIZE(tap_dance_actions) <= (QK_TAP_DANCE_MAX - QK_TAP_DANCE), "Number of tap dance actions exceeds maximum. Are you using SAFE_RANGE in tap dance enum?");
      |                ^~~~~~~~~~
quantum/keymap_introspection.c: In function 'tap_dance_count_raw':
quantum/keymap_introspection.c:130:1: error: control reaches end of non-void function [-Werror=return-type]
  130 | }
      | ^
quantum/keymap_introspection.c: In function 'tap_dance_get_raw':
quantum/keymap_introspection.c:143:1: error: control reaches end of non-void function [-Werror=return-type]
  143 | }
      | ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make: *** [builddefs/common_rules.mk:362: .build/obj_geigeigeist_totem_vial/quantum/keymap_introspection.o] Error 1
➜  totem git:(master) ✗
```
unless I commented out `KEY_OVERRIDE_ENABLE = yes` in `keymaps/vial/rules.mk`. Not sure if that’s expected behavior or something that needs attention.

After flashing, I wasn't able to configure the keymap in usevia.app or the Vial software. Vial gave me this error in a prompt:

> Unsupported protocol version!
> Please download latest Vial from [https://get.vial.today/](https://get.vial.today/)

Also, under `About > Geist Totem` in Vial, I get the following traceback:

```
File "main_window.py", line 433, in about_keyboard
  File "about_keyboard.py", line 82, in __init__
  File "about_keyboard.py", line 20, in about_tap_dance
AttributeError: 'Keyboard' object has no attribute 'tap_dance_count'
```

In VIA, the configurator shows the layout after loading the JSON from the design tab, but the Configuration tab is empty. I also had to set the version to v2 (deprecated) for the `vial.json` just to get it to load.

PS: I now tried to get it to run since two whole days idk what to do anymore i rewrote the vial.json added an info.json to the root dir modified the rules over and over again. Nothing worked.

Kitiar, 2025-06-09:
Hello, I have tried to use your recomendations about 
* `SPLIT_KEYBOARD = yes`
* `KEY_OVERRIDE_ENABLE = no`
But buit firmware doesn't reflect my layout. Looks like qmk uses somthing default layout instead of mine.

I am using the wired version of the Totem.
My layout configuration is https://github.com/Kitiar/TOTEM/tree/main/firmware/QMK/VIAL_source/totem/keymaps/kit

transporsche, 2025-08-09:
Greetings, i have successfully compiled your firmware. but i only get one uf2 file :( shouldnt it get two for left and right`?

transporsche, 2025-08-26:
> > Greetings, i have successfully compiled your firmware. but i only get one uf2 file :( shouldnt it get two for left and right`?
> 
> Did you figure out why it creates only 1 uf2?

Greetings, yes you have to flash both sides with the same file.

klytuko, 2025-08-26:
> > > Greetings, i have successfully compiled your firmware. but i only get one uf2 file :( shouldnt it get two for left and right`?
> > 
> > 
> > Did you figure out why it creates only 1 uf2?
> 
> Greetings, yes you have to flash both sides with the same file.

Yep, thought that was the right procedure however doesn't seem to function for me. The original geist left and right uf2 works but with these "single file" ones I've noticed that the right side just doesn't seem to function at all. Workaround is just using something like vial.rocks and mapping there. First split keyboard project so I'm awful with tshooting these problems.