GEIGEIGEIST/TOTEM #61: What exactly is outdated with the source firmware?

iamavolk, 2025-06-23:
First of all, great project, many thanks! 

Back to the topic, though: just wanted to see what aspects of the source repo/guide are outdated (per the readme on here), as I couldn't really find anything specific upon perusing the repo itself?

Ratione, 2025-07-06:
So, Is everyone who is attempting to build the wired version just waiting until someone figures out how to make this work with the new QMK version, or do we need to look for alternative solutions, or what?

Momro, 2025-07-30:
Hi all,

I received a current qmk firmware from a friend who has a working config. I was able to compile it with current Ubuntu 24.04 in wsl with qmk vial.

Now I'm trying to publish the config and everything to github. 

I forked the repo https://github.com/Momro/Totem

I also uploaded my newbie explanation how to compile the qmk-vial firmware in this readme 

https://github.com/Momro/Totem/tree/main/firmware/QMK/VIAL_source/totem

I hope to upload my current firmware there as well. Hopefully tonight. I have to perform a few tests first.

iamavolk, 2025-08-15:
This is the ZMK config. The original structure was a bit off, so I updated it. Works fine.

Clone the repo and just write your own config in 
totem.keymap, commit, push, and flash the produced firmware.

https://github.com/iamavolk/totem-shield-module